Harvey71/emdr #1: Einschraubmuffe

LennartHennigs, 2021-05-01:
Hey Thomas, 
super Projekt - ich frage mal auf Deutsch, da mich die deutschen Maße interessieren: Was für eine Muffengröße hast du für das 3/8" Gewinde genommen? Ich muss ja grade im Baumarkt vorbestellen und es leider nicht vor Ort ausprobieren. 

Ich baue auch grade eine EMDR Leiste und habe zum Glück dein Video gefunden. 

(Und über die "Massage" Eier habe ich sehr lachen müssen. Großartige Idee.)

VG
Lennart

Harvey71, 2021-05-01:
Hallo Lennart,
ich habe bei Amazon einen 1/4 auf 3/8-Zoll Adapter bestellt (https://www.amazon.de/gp/product/B087JTZ6XJ).
Eigentlich ist der dazu da, in Kameras mit 3/8-Zoll-Gewinde eingedreht zu werden, um diese dann auf ein Stativ mit 1/4-Zoll-Gewinde zu montieren. Das Außengewinde ist ca. 7,3mm (Kernloch) auf 9,0mm (Außendurchmesser) und das habe ich als selbstschneidendes Gewinde missbraucht.
Dazu habe ich mit einem 8mm-Bohrer ein Senkloch mittig in die Holzleiste gebohrt. Da der Adapter auch noch einen Kragen von 9,5mm Duchmesser und 1,5mm Dicke hat, habe ich das Loch 1,5mm tief mit einem 10mm-Holzbohrer aufgeweitet. Es bedarf etwas Kraft, Gedult und einen passenden Schlitzschraubendreher, um den Adapter dann in dieses Senkloch einzudrehen. Dafür sitzt das dann aber auch fest und bietet ein sauberes 1/4-Zoll Innengewinde für das Stativ. 
Am besten siehst du, wie ich das Gemacht habe im Video (https://www.youtube.com/watch?v=0NRr-7e_mLk) von 0:30 bis 1:15 min.
VG
Thomas

LennartHennigs, 2021-05-01:
Ah, super. Danke für die schnelle Antwort, das hilft mir weiter. Den Adapter habe ich auch. Dann werde ich noch mal in Ruhe Dein Video schauen. 
Vielen Dank.

closed 2021-06-24 by LennartHennigs