Harvey71/emdr #4: Pibakery not available

dmbe11, 2023-01-23:
Do you know of another way to build the image for the lightbar controller without the pibakery. It seems to be gone and the old versions wont work

thx

Harvey71, 2023-01-23:
You can still install the Pi the classic way:

1. Flash a recent Rasbian image (desktop version) and boot the Pi
2. Connect Pi to you local LAN
3. Log in with SSH
4. Install libraries: 
```
sudo apt-get update
sudo apt-get install libusb-1.0.0-dev
sudo apt-get install python3-sdl
```
5. Clone repo and install python packages:
```
git clone https://github.com/Harvey71/emdr
cp emdr/controller/* ~ -r
sudo pip3 install -r requirements.txt
```
6. Configure auto-start
```
mkdir -p ~/.config/lxsession/LXDE-pi
echo sudo python3 main.py fullscreen touchscreen >~/.config/lxsession/LXDE-pi/autostart
```
7. Install and config LCD driver
```
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show
./MHS35-show 180
```
8. Configure audio
```
echo "dtparam=audio=on"|cat - boot/config.txt &>/boot/config.tmp
mv /boot/config.tmp /boot/config.txt
```
9. Reboot
10. Now, the LCD should show the controller GUI

Unfortunately, I do not have a spare Raspberry Pi at hand, so I cannot test the above instructions. Maybe you have to adjust a thing or two, depending on Rapsbian version.

Good luck!

dmbe11, 2023-01-24:
Thanks, I thought that would be the next option. Thank you for verifying. The ability to give people suffering from trauma is really dramatic with these types of devices.
Thanks for putting this out here.

dmbe11, 2023-01-24:
I was able to install everything except for the Python3-sdl but I was able to install the sdl2 version that I found as a solution. I ran the print (dev) script you posted and received this:
DEVICE ID 1d6b:0003 on Bus 002 Address 001 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x300 USB 3.0
 bDeviceClass           :    0x9 Hub
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x3
 bMaxPacketSize0        :    0x9 (9 bytes)
 idVendor               : 0x1d6b
 idProduct              : 0x0003
 bcdDevice              :  0x515 Device 5.15
 iManufacturer          :    0x3 Error Accessing String
 iProduct               :    0x2 Error Accessing String
 iSerialNumber          :    0x1 Error Accessing String
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 0 mA ====================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x1f (31 bytes)
   bNumInterfaces       :    0x1
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0 
   bmAttributes         :   0xe0 Self Powered, Remote Wakeup
   bMaxPower            :    0x0 (0 mA)
    INTERFACE 0: Hub =======================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x9 Hub
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x4 (4 bytes)
       bInterval        :    0xc
DEVICE ID 045e:0800 on Bus 001 Address 003 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x045e
 idProduct              : 0x0800
 bcdDevice              :  0x934 Device 9.34
 iManufacturer          :    0x1 Error Accessing String
 iProduct               :    0x2 Error Accessing String
 iSerialNumber          :    0x0 
 bNumConfigurations     :    0x1
  CONFIGURATION 1: 100 mA ==================================
   bLength              :    0x9 (9 bytes)
   bDescriptorType      :    0x2 Configuration
   wTotalLength         :   0x54 (84 bytes)
   bNumInterfaces       :    0x3
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0 
   bmAttributes         :   0xa0 Bus Powered, Remote Wakeup
   bMaxPower            :   0x32 (100 mA)
    INTERFACE 0: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
[... 108 lines cut ...]
   bConfigurationValue  :    0x1
   iConfiguration       :    0x0 
   bmAttributes         :   0xe0 Self Powered, Remote Wakeup
   bMaxPower            :    0x0 (0 mA)
    INTERFACE 0: Hub =======================================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x9 Hub
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0 
      ENDPOINT 0x81: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x81 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :    0x4 (4 bytes)
       bInterval        :    0xc
>>> 
Im not sure what I am looking. Here is some additional info that might help:

the GUI loads but is is pretty small. When I get the buttons to work I can once in a while get a single led to work. If I run one of the sketches from the  WS2812 Serial library and they work fine. The Only other thing I can think of is that I have the 2812b lights. Anything come to mind. The behavior is really close to what I got when running the controller via my mac or another machine I tested on that is a linux mint PC.  I'm kind of stuck since I don't really understand the program I think because it is not really operated from the sketch on the teensy but somewhere in the python files.  thx again

dmbe11, 2023-01-24:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Thomas,
I had ordered the correct size screen and it came in today.  I installed and it loads in the pi just fine. It still does not control the led strip and I am not sure why. Any ideas? Could it be the strip is the B version of the 2812’s: 2812B?

  1.  Clone repo and install python packages:

  1.  Configure auto-start

  1.  Install and config LCD driver

  1.  Configure audio

  1.  Reboot
  2.  Now, the LCD should show the controller GUI

closed 2023-01-25 by dmbe11