Harvey71/emdr #5: Tinsy lc

MrLately, 2023-01-27:
Hello Harvey, thank you for this great tool. I'm having trouble finding timsy lc. What are some compatible devices if some of these aren't available?

Harvey71, 2023-01-27:
Hi Mr Lately, I just took a look and I am shocked by the prices you have to pay for a Teensy LC nowadays - if you can get one at all. Sadly, I don't think that there is a drop-in replacement, but actually you can use any microcontroller with an active USB port. If I would do this project now, I would choose a RP2040 based µC, like the Raspberry Pico. I chose the Teensy LC, because it has a 5V driver on board. But it seems that actual WS2812B (with the B) LED stips work fine with 3.3V output of a Pico - but no guarantees. Please refer to this link: https://www.freva.com/how-to-control-a-neopixel-led-strip-with-a-raspberry-pi-pico/ I am pretty sure, you have to adjust the source code for the lightbar firmware. I am sorry, but I currently don't have the time (and equipment) to do that, so that would be up to you. Maybe I will find the time later this year. I will order a LED strip right away.

MrLately, 2023-01-27:
I just bought the ws2812b / 60 led / 1 meter strip. I've written a tkinter script with a canvas already then I found your project. Was thinking about using a pico or just run it from a 3b with the rpi_281x library to minimize hardware. I will put what I have on my GitHub later if you want to take a look in your free time. Thank you for your response and time.

MrLately, 2023-01-31:
https://github.com/MrLately/blink-and-slide/tree/main

closed 2023-04-21 by Harvey71