Harvey71/emdr #6: Updating problems getting light bar to work.

dmbe11, 2023-02-01:
Thomas,
I was out of town for a bit and didn't have a chance to reply. My issue is not with the controller; it is working. The issue is either with the sketch or the communication between the controller and the Teensy. 

Here are some clues that I observed:
1. The sound works from the controller, but I can only get a single led to the light.  
2. I get the same responses if I run the controller from a Mac, Mint Linux, or the pie.
3. When I say the controller is working, I mean the play, pause, and stop buttons. The eye icon and the? Done seen to have any effect.
Also, do you have the procedure you posted for loading the controller on the pi? I may need to reinstall it, and I did not save it anywhere. I hope this helps; if not perhaps you could suggest where to start looking>
thank you again

Harvey71, 2023-02-05:
When you execute this 
```
import usb
for dev in usb.core.find(find_all=True):
    print(dev)
```
in your Python environment with properly flashed and attached Teensy LC, there should be something like this
```
DEVICE ID 16c0:0483 on Bus 020 Address 022 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x110 USB 1.1
 bDeviceClass           :    0x2 Communications Device
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x16c0
 idProduct              : 0x0483
 bcdDevice              :  0x273 Device 2.73
 iManufacturer          :    0x1 Teensyduino
 iProduct               :    0x2 USB Serial
 iSerialNumber          :    0x3 5667580
 bNumConfigurations     :    0x1
```
printed to screen. If not, your Teensy is either not properly flashed or not properly attached to your USB. To check that, you do **not** need the controller software:

1. Check, whether Teensy is properly flashed.
When attaching the light bar to your USB port, the outer LEDS should be lit for half a second, one in red, the other in green. If you see only the green one, you have to adjust NUMLED in the firmware's source code to the number of LEDs of your stripe. If you do not see the 0.5 secs flash at all, something went wrong with programming the Teensy. Try it again.
2. Try another USB cable.
Surprisingly, not all USB cables work properly with Teensy. So best try some different vendors and lengths. I got best results with Amazon Basic Cables.

Good luck

dmbe11, 2023-02-06:
Thank you for the information. I verified that the teensy is working correctly, but unfortunately, I had to reinstall the software on my pi. I decided to install the legacy 32-bit version and would like you to provide the steps for installing the Controller Software on the PI. Here is what I recall stepwise:
1. Create a Controller Directory
2. download EMDR folders and files
3. pip3 install -r requirements.txt
4. Download and install libusb
5. Setup autostart on pi

That's about what I recall. Thanks for your help. 

Dan

By the way! Harvey is a great film. I have a copy on DVD that I dug out a week ago. The lady who played the female nurse was a good actress but retired only a few years later to marry. They stayed married for 60+ years, quite uncommon for an actor today.

Harvey71, 2023-02-06:
For setting up Raspberry Pi without PiBakery, please refer to: https://github.com/Harvey71/emdr/issues/4

dmbe11, 2023-02-06:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Thank you, Thomas

Daniel Bell, PhD
Program Chair, Interactive Media Design
Franklin University
***@***.******@***.***>
614.947.6117

dmbe11, 2023-02-07:
Hi Thomas,
any idea why this command is returning Permission Denied?

echo "dtparam=audio=on"|cat - boot/config.txt &>/boot/config.tmp

Harvey71, 2023-02-07:
You have to be root to write to that file. Try this before:
`sudo su`

dmbe11, 2023-02-07:
would python3-sdl2 be a viable alternative for E: python3-sdl?  I get an error: E: Unable to locate package python3-sdl

Harvey71, 2023-02-07:
Should work, too. Just try it. :-)

closed 2023-04-21 by Harvey71