Harvey71/emdr #7: Very cool design! Some hook up questions about pin 17 driver

ooktown, 2023-03-31:
Greetings! First: THANK YOU for open-sourcing the design of this EMDR kit!

The Teensy LC's are hard to come by, so I am planning on using Teensy 4.0 with a TTL level shifter to go from 3.3V to 5V.

My question: Will that work?

Can I then hook up pin 24 of the Teensy 4.0 via the level shifter to the DI of the WS2812?

Harvey71, 2023-04-01:
Teensy 4.0 with level shifter should work fine. You can use any pin that is supported by the WS2812Serial library - this includes pin 24. You could even try without level shifter. Maybe you are lucky, but I would not recommend it. WS2812B datasheet claims 3.5V is minimum for logic high, so 3.3V is too low. If you use a level shifter, please use a fast one (e.g. TXS0108E). I read about troubles with simple bidirectional level shifters for I2C, since they skew too slow from low to high.

ooktown, 2023-04-01:
Thank you! That answers my question. Now I just need to wait for RPi4 to become reasonably priced... In the meantime I'll put together the kit over time and try it out with my Mac.

closed 2023-04-01 by ooktown