Harvey71/emdr #9: Audio

deestruct, 2023-07-03:
After struggling to get the setup to function from the tutorials, I was finally able to get it working using the instructions provided in [](https://github.com/Harvey71/emdr/issues/3#issuecomment-1517065070).  I am still awaiting some components to arrive so have not completed the fabricating the buzzers and lightbar hardware devices, but did get the RPi 4 and TFT 3.5" screen working now.

I am not getting audio though from the device on my headset connected to the 3.5mm jack when I test the sound via the GUI.  I can confirm that the audio is working in the RPi as when I connect to it via SSH and perform a basic audio test with the code `speaker-test -c2 -twav -l7`, I can hear the test functioning properly in my headset.  Immediately after the audio test is complete, I try the audio again using the GUI on the TFT screen and I cannot hear any sounds.

However, if I connect the RPi to my monitor using an HDMI cable and try testing the audio on the EMDR GUI I can hear the tones properly.

Can you please assist me in resolving this issue?  I am stumped why the only element not working in the audio output is the headphone jack via the EMDR GUI audio test.

Harvey71, 2023-07-03:
Hi deestruct, please try to configure default audio output to jack. One way to do so is via command line:
`sudo amixer cset numid=3 1`

deestruct, 2023-07-04:
I tried the suggested command via SSH and it did not work either.

deestruct, 2023-07-04:
Perhaps I am doing something wrong in testing the audio via headphones.  Below are the steps I am performing to test the headphone jack for audible tones; I performed the steps below today after entering the command line you suggested and rebooting the system.

1) Press the EAR icon to enter the sound menu:
[image]

2) Try the vertical bar icon and then press the '?' to test the sound:
[image]

3) Try the circular icon and then press the '?' to test the sound:
[image]

Harvey71, 2023-07-05:
I found something else in rhe PiBakery recipy that may help:

`echo "dtparam=audio=on"|cat - /boot/config.txt >/boot/config.tmp&&mv /boot/config.tmp /boot/config.txt`

You would have to execute this as root and do a reboot.

deestruct, 2023-07-05:
I didn't initially execute that command as when I checked the config file it already had the following lines active:
[image]

I did execute the command though and it added the same line at the top of the config file as shown below:
[image]

After rebooting, it still behaves the same.

closed 2023-09-02 by deestruct