Harvey71/emdr #13: Issue when Stoping

KaboomFox, 2024-06-14:
When I press the stop button, The buzzer stops. However, light bar slows down but still moves. It then will not allow me to press play again. It is still greyed out. It never recovers.

Harvey71, 2024-06-16:
Could you give me some more information, like
- Do you use the latest controller software version from this repository?
- On what system (Mac, Windows PC, Raspberry Pi) are you running the controller app?
- How many LEDs do you have? Have you set this value both in the controller app (`num_led` devices.py) and light bar firmware (`NUMLED` in lightbar/lightbar.ino or lightbar/main.py)?
- Could you give me the output of `pip freeze`?
- You say, the light bar slows down. That's intentional. But it should stop at the middle of the light bar. Does it do that or will it move forever?
- Could you please take either a screenshot of the stuck application or describe exactly which buttons are greyed out?

KaboomFox, 2024-06-19:
I'm using the latest controller software. I'm using a Raspberry PI 5, 7" raspberry pi touchscreen and two Teensy LCs . I am also using https://www.amazon.com/gp/product/B08XXBSF1P/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1 for vibration motors but I don't think that is the issue.

I have 98 LEDs. I changed the number in both files. I have a 7" screen and if I go above a certain screen size, the light bar and vibration pads go out of sync or I get a seg fault. I can avoid this issue by making the screen size slightly smaller than fullscreen.

Once I press stop the light bar slows down and slowly goes to the middle of the bar but it takes 2 minutes to do so. I think the issue is the number of LEDs. I changed the main.py file back to 57 and it worked better. Any suggestions? Trying to avoid remaking the light bar.
[image]

KaboomFox, 2024-06-19:
Also thank you! This project is amazing.

Harvey71, 2024-06-20:
The reason for the issue was the higher number of LEDs. The speed decay was only tuned and tested for 57 LEDs and way too slow for 98. I changed the calculation and it will now always decay in a way that the delay between the last two LEDs will be 1.5 seconds, exponentially approximated. If you update to the latest version or just cherry-pick commit 4f339f9d6e1e9598fdfcbe8978ded8839cbd7c1d your issue should be fixed.

If you want to change the behaviour, you can edit line 580 of main.py - best change the "1.5" to any amount of seconds you want the last delay to be.

KaboomFox, 2024-06-24:
Related issue:

At higher speeds the vibration pads get out of sync and I get an interesting freezing with the lights. Any ideas?

Uploading IMG_4919.MOV…

Harvey71, 2024-06-29:
This issue should be fixed now, too. Please update all firmware (light bar and buzzers) and controller software.

KaboomFox, 2024-07-12:
Unfortunately, now the light doesn't go across the whole bar. I updated` led_num` in devices.py to`98` and `NUMLED` to `98` in the lightbar.ino file.
https://github.com/user-attachments/assets/c3694390-6b16-40ed-9045-93ea73af7f13

Harvey71, 2024-07-12:
It should work with 100 LEDs though. Before you switch them out, could you please do the following checks?
- Disconnect and reconnect the light bar. Does the left most LED light up green and the right most LED light up red? If not, please double check the INO file and burn the firmware again.
- On the controller start the light bar test mode. You can do so by clicking on the icon with the eye and then click on the icon with the questionmark. Does the left most and right most LED light up? If not, please double check your devices.py file and copy the Python files to the Raspberry Pi again.

Harvey71, 2024-07-12:
BTW: I have ordered a 1 M 100 LED stripe, which will arrive tomorrow. I will test it as soon as I hold it in my hands and come back to you.

Harvey71, 2024-07-13:
OMG. I found the bug. In one place in the lightbar.ino file, there was still a hard coded limit of 60 LEDs. 😳 Fixed with the latest commit. You would only have to update the lightbar firmware. Sorry for the inconvenience.

KaboomFox, 2024-07-14:
You are awesome. Looks like everything is working. I'm going to run some tests and make sure nothing gets out of sync during a 90 min session but looking good so far.

KaboomFox, 2024-08-01:
Closing the ticket. Thank you!

closed 2024-08-01 by KaboomFox