Harvey71/emdr #15: How to implement ESP32

Aaryan-Kapoor, 2025-02-12:
I tried going through and changing stuff but wasn't able to get it to work with my ESP32.

Harvey71, 2025-02-12:
Hey, @Aaryan-Kapoor.
Could you provide more information? What exactly is not working with ESP32?
Depending on your answer, other pieces of information could be relevant, like:
- How many LEDs has your LED strip?
- Where is your controller software running (Mac, Windows, Linux, Raspberry Pi)?
- When connecting the ESP, does the light bar show the init colors (red and green at the end of the light bar)?
- Can the host system, where the controller runs on, see the USB devices (try lsudb)?
- What did you change in the code apart from the number of LEDs?

Aaryan-Kapoor, 2025-02-12:
@Harvey71 Apologies for the vague message earlier. Here are the details of my setup and the issue I'm facing:

### System & Hardware Details:
- **Operating System**: Windows  
- **Microcontroller**: ESP32-WROOM ([Amazon](https://a.co/d/iSzeHEn))  
- **LED Strip**: Cut to 57 LEDs ([Amazon](https://a.co/d/4YcItI6))
- **MicroPython Installed**: ESP32 appears under COM6  

- [image]

### Issue:
- The **initial red/green lights** do not turn on when connecting the ESP32.  
- The LED strip itself **works fine** when tested with a Raspberry Pi Pico.  
- **ESP32 Setup:**
  - 5V connected to **VIN**
  - GND connected to **GND**
  - Data line connected to **RX2** (I tried different pins but nothing seemed to work)

### Code Modifications:
I've updated the number of LEDs in the following files:  
**Controller directory:**
- `devices.py`: Changed `led_num` to **57**  

**Firmware directory:**
- `lightbar.ino`: Changed `NUMLED` to **57**  
- `main.py`: Changed `NUMLED` to **57**  

### Troubleshooting Steps Taken:
- **Tried using a level shifter** in the middle replicating the D1 ESP8266 mini setup in this repo, but the issue persists.  
- The same LED strip **worked with [WLED](https://kno.wled.ge/)** without needing a level shifter.  

### ESP32 Pin Layout:
```
ROW1: VIN, GND, D13, D12, D14, D27, D26, D25, D33, D32, D35, D34, UN, UP, EN
ROW2: 3V3, GND, D15, D2, D4, RX2, TX2, D5, D18, D19, D21, RX0, TX0, D22, D23
```

### Suspected Issues:
1. **Incorrect pin connections**  
2. **Project may not support ESP32** (ESP32 is not listed in `controller/devices_config.py`)  

Let me know if you need any additional details! I would really appreciate your help!

Aaryan-Kapoor, 2025-02-19:
Hey @Harvey71,

I would really appreciate your help with this project and am looking forward to hearing from you soon!

Harvey71, 2025-02-21:
Sorry, Aayan. Actually, the ESP32-Wroom was not supported by this project. 
But I ordered one to try it out. I just committed a change in the firmware code to support your microcontroller. 

Please follow the D1-Mini instructions, **except**:
- Connect the data line to **D4** on the ESP developer board.
- I recommend using a level shifter and have not tested without it, but if it works for you, just go for it.
- At "Configure Interpreter" use "**MyrcoPython (ESP32)**" as interpreter, "**ESP32**" as family, "**Espressif - ESP32 / WROOM**" as variant and "**1.24.1**" as version.
- Update your git repo, adjust your LED num and copy "main.py" and "neopixel.py" to your microcontroller
- After reboot, your outer LED should light up in red and green for a second.

Please let me know whether this solves your issue.

Aaryan-Kapoor, 2025-02-22:
Hey @Harvey71, thank you so much for the response!

So progress is that the initialization lights now light up but the controller software is not able to detect the ESP32. Thonny detects it and I am able to upload new code.

I tested both with and without the level shifter, but still had the same behavior.

Harvey71, 2025-02-22:
That the initialization lights work is a good sign. But I forgot to adjust the controller software. I I have just made up for it. Please pull and replace the controller source code (at least the file "device_setup.py").